frg-fossee/eSim-Cloud #7: 'ng serve' exits with error

darshkpatel, 2020-04-25:
I was trying to setup the dev environment for ```ArduinoFrontend``` 
but ```ng serve``` exits with the following error.
```Invalid "exports" main target "index.js" defined in the package config /Users/darsh/github/eSim-Cloud/ArduinoFrontend/node_modules/is-promise/package.json```

Steps followed: 
1. ``` npm install ```
2. ``` ng serve ```

NavonilDas, 2020-04-25:
This Error is because of the node js version you are using try to downgrade the node js version.

NavonilDas, 2020-04-26:
Apparently this bug is caused in the is-promise library. @meet-10 also got the same error we fixed it after reading [this thread](https://github.com/facebook/create-react-app/issues/8896).

darshkpatel, 2020-04-26:
Ref: #9 
Sent a PR with a container with the appropriate node version.

closed 2020-04-26 by darshkpatel