frg-fossee/eSim-Cloud #46: Failing Build eda-frontend

darshkpatel, 2020-05-15:
Build is failing due to linting issues in files 
./src/components/SchematicEditor/Helper/ComponentDrag.js
./src/components/SchematicEditor/Helper/SideBar.js

darshkpatel, 2020-05-15:
@felixfaisal

darshkpatel, 2020-05-15:
./src/components/SchematicEditor/Helper/xml_parser.js is also throwing linting errors @rohitgeddam

darshkpatel, 2020-05-18:
In the latest run , the tests are failing due to the hardcoded svg imports, 
https://github.com/frg-fossee/eSim-Cloud/runs/684106962

darshkpatel, 2020-05-18:
```

      1 | import mxGraphFactory from "mxgraph";
    > 2 | import comp1 from "../../../static/CircuitComp/U-4001-1:A.svg";
        | ^
```
``` src/components/SchematicEditor/Helper/ComponentDrag.js:2:1 ```

dssudake, 2020-05-18:
@darshkpatel I have fixed issue. Pushed latest changes on eda-toolbar.
ref: [ddec2d1](https://github.com/frg-fossee/eSim-Cloud/commit/ddec2d197423fbbd0fc90ea6a2e0e22314ec529f)
https://github.com/frg-fossee/eSim-Cloud/runs/684448462

closed 2020-05-18 by dssudake