frg-fossee/eSim-Cloud #78: arduino path improperly configured

darshkpatel, 2020-05-27:
```http://localhost/arduino/```
Base href environment variable does not seem to affect the development build paths and still tries to load assets from ``` / ```

[image]

darshkpatel, 2020-05-27:
same thing happens for the production config , built js files are fine , but the paths are not affected by changing the base href 
```http://localhost/assets/images/arduino.JPG```
it should be loading images from ``` http://localhost/arduino/assets/images/arduino.JPG```
must be a setting in angular build 
@NavonilDas  can you have a look

NavonilDas, 2020-05-27:
@darshkpatel The Base Href should work.
In the build config, we only need to mention the assets directory it basically copies the assets directory to build directory.

I guess we can do one thing add a /assets route in NGINX Config. It will also help us cache images & js files.

darshkpatel, 2020-05-27:
All images are already being cached globally so don't have to define a different route for it. 

Updating the angular config and changing the base url properly might be a better approach than reserving the assets route. 

For react it's simply setting the host url checkout the env file

NavonilDas, 2020-05-27:
@darshkpatel the build configs are correct. I guess on **"../assets"** needs to be updated as **"./assets"** on the dashboard. Can you open the simulator **localhost/arduino/simulator** and see if the images on components are loading properly.

darshkpatel, 2020-05-27:
> @darshkpatel the build configs are correct. I guess **"../assets"** needs to be updated as **"./assets"** on the dashboard. Can you open the simulator **localhost/arduino/simulator** and see if the images on components are loading properly.

[image]

darshkpatel, 2020-05-29:
https://github.com/frg-fossee/eSim-Cloud/blob/966db7d0d8ac448d97f99e2199e41a009d47ecee/Nginx/conf.d/local.conf#L62

That snipped does not seem to be able to fix the issue , i cant seem to figure it out. Any ideas ?

darshkpatel, 2020-05-30:
[image]

@NavonilDas It's still loading stuff from /assets

closed 2020-06-06 by NavonilDas