frg-fossee/eSim-Cloud #168: Adding LCD Support

VishnuEaswaran, 2020-07-24:
The current version have a dummy LCD in place but lacks functionality. Let us built an emulator for the same to make it useful.

Some references:
- The LiquidCrystal Arduino library source: https://github.com/arduino-libraries/LiquidCrystal
- de-facto choice of LCD driver: https://en.wikipedia.org/wiki/Hitachi_HD44780_LCD_controller

The name of the popular LCD module is LCD1602 and internally it houses the Hitachi driver.

[image]

VishnuEaswaran, 2020-07-25:
An LCD 2*16 simulator: http://www.dinceraydin.com/djlcdsim/djlcdsim.html

VishnuEaswaran, 2020-08-03:
[image]
towards the bottom of the page, we have a download option which seems to be of the source code. Said that, there is no license in the download.

firuza, 2020-08-06:
https://github.com/visrealm/vrEmuLcd

firuza, 2020-09-01:
Implemented by gupta-arpit. Feature merged in develop branch.

closed 2020-09-01 by firuza