frg-fossee/eSim-Cloud #171: UX: Pressing escape should delete the connecting line in progress

gupta-arpit, 2020-07-31:
- If you're in the process of connecting nodes, pressing `esc` should delete the in-progress line.

[image]

NavonilDas, 2020-08-01:
To remove existing wire press **Delete**  Key.

If you want to change the key to **ESC** Update the following Line in [Workspace.ts](https://github.com/frg-fossee/eSim-Cloud/blob/master/ArduinoFrontend/src/app/Libs/Workspace.ts#L423).

gupta-arpit, 2020-08-03:
I am using macOS, I tried pressing the delete button. It doesn't work

NavonilDas, 2020-08-04:
> I am using macOS, I tried pressing the delete button. It doesn't work

Not Sure about mac but it works perfectly on windows.

firuza, 2020-08-17:
Implemented by @gupta-arpit in #178

closed 2020-08-17 by firuza