frg-fossee/eSim-Cloud #192: LED/Buzzer issue

firuza, 2020-09-13:
If the 'negative' of LED is connected to pin 13 of Arduino using breadboard and positive pin of LED is not connected (basically 0) connected anywhere on breadboard, even then the LED glows. This can be checked by connecting using wires (one leg at 0 and another at 5V).

Ideally, the LED should not glow. This is true for buzzer too.

AadishKapoor25, 2020-09-24:
Connect the Negative terminal of LED to GND of Arduino board and Positive terminal of LED to any of the GPIO pins.

Ohhmydude, 2021-05-15:
[image]

Abcdma, 2022-01-27:
> If the 'negative' of LED is connected to pin 13 of Arduino using breadboard and positive pin of LED is not connected (basically 0) connected anywhere on breadboard, even then the LED glows. This can be checked by connecting using wires (one leg at 0 and another at 5V).

I think you should connect the negative terminal of LED to GND of Arduino board and Positive to any GPIO pins.

firuza, 2023-03-03:
Fixed in #467