frg-fossee/eSim-Cloud #243: Netlist corresponding to equivalent circuits differ in case of supernodes when the points of connections are altered in eSim

Kaustuv942, 2021-05-20:
In a situation where multiple points are connected through a supernode involving (say) three components C1, R1, and Q1, it was observed when the connection of the three components was made at the R1 node resulted in a different netlist than when the same connection was made at C1 node. 

In effect, both the circuits are equivalent and should result in the same graph. This is a case of an error in the generation of the netlist.

[image]

felixfaisal, 2021-05-20:
Hey, @Kaustuv942 there's an option to view the netlist, So if you could check the netlist before and after you made the change then you can better analyze where the problem lies.

firuza, 2021-08-01:
Fixed in #336

closed 2021-08-01 by firuza