frg-fossee/eSim-Cloud #471: Buzzer connection issue

rinon13, 2023-02-17:
Description:Buzzer should only work when the POSITIVE pin is HIGH and the NEGATIVE pin is LOW. In the scenarios shown below, the Buzzer is not following that principle. 

POSITIVE pin is the left pin of the Buzzer and the NEGATIVE is the right pin.

[image]

[image]

Scenario 1 : Buzzer works when POSITIVE pin is connected to GND (0/LOW) and NEGATIVE pin is connected to pin 13 of Arduino (programmed as HIGH).

[image]

Scenario 2 : Buzzer works when POSITIVE pin is connected to pin 13 of Arduino (programmed HIGH) and NEGATIVE pin is connected to any unassigned/unused pin of Arduino. (In figure below, pin 3 is not assigned in the .ino code)

[image]

Scenario 3: Buzzer works when POSITIVE is connected to pin 13 of Arduino (programmed HIGH) and NEGATIVE pin is connected to unconnected Resistor (or any other component) and vice versa.

[image]

[image]

Steps to Reproduce:

1. Connect circuit as shown in figures above.
2. Copy .ino code.
```
void setup(){
	pinMode(13, OUTPUT);
}

void loop(){
	digitalWrite(13, HIGH);
    delay(1000);
	digitalWrite(13, LOW);
    delay(1000);
}
```
3. Press "Start Simulation"
4. Buzzer will work.

Expected behavior: Buzzer should not work and throw an error.

 - OS: Ubuntu 22.04.1 LTS 
 - Browser: Firefox 110 (Ubuntu)

Tested on:
- develop
- simulation.iitbx.in

RAKSHIT-RAJ2028, 2023-04-27:
Assign: @RAKSHIT-RAJ2028

RAKSHIT-RAJ2028, 2023-04-27:
can i do this task?

Eyantra698Sumanto, 2023-04-27:
@RAKSHIT-RAJ2028 you may try it out and  send your updated GitHub branch here and/or create a Pull request.