frg-fossee/eSim-Cloud #476: Arduinos parallel execution is not in sync

rinon13, 2023-02-21:
Description: When 2 Arduinos connected to LEDs are executed in parallel with the same .ino code, the LED are not blinking at the same time.

Steps to Reproduce

[image]

1. Connect as shown above or take circuit from Gallery (Two Arduinos)
2. Add .ino code to both Arduinos
```
void setup(){
	pinMode(LED_BUILTIN, OUTPUT);
}

void loop(){
	digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
	digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
}
```

3. Press "Start Simulation"
4. LEDs are not blinking in sync

Video of the process mentioned above.
[Two_Ardinos_Problem.webm](https://user-images.githubusercontent.com/87307364/220279874-14a1acb6-1c61-45a2-a808-cf9e1eb0d6e6.webm)

Expected Behavior: Both LEDs are supposed to be blinking in sync.

Version and OS:
 - OS: Ubuntu 22.04.1 LTS 
 - Browser: Firefox 110

Tested on:
 - develop
 - simulation.iitbx.in

firuza, 2023-03-04:
Fixed in #479

firuza, 2023-03-04:
Fixed in #479

closed 2023-03-04 by firuza